appidea/react-native-hce #1: How can i know the data which i want to share is scanned by other Device?

shashi-kiwi, 2020-12-19:
Hi,
 I start simulation and data is scanned by other device but How can i get to know that the other device has scanned?
Thanks in Advance . Please answer asap.
Thanks,
Shashi

spsingh559, 2021-04-25:
I am having a similar question. Can we implement HCE reader in react native? All I can see is Emulating an NFC Card using React native but what about the reader app which can read emulated device?

My use case is to send a personal identifiable information data from one app to other in different mobile phones with NFC enabled. It would be great to see if any peer-to-peer data sharing using NFC example is there.

appidea, 2021-08-13:
You probably need to use this library combined with react-native-nfc-manager to reach this kind of functionality

closed 2021-08-13 by appidea

SurpriseMF3000, 2022-09-01:
Any new method?