appidea/react-native-hce #2: The Data is sent only which App is in Background?

shashi-kiwi, 2020-12-19:
Hi,
 I integrate this lib in my current project but the issue is when i start simulation with text, if app is in forground the other device don't scan text. But If i put app in background then Other device recieve data .
Please help in doing this in foreground as well.
Thanks in Advance.Please answer asap.
Shashi.

vasilycrespo, 2022-08-01:
Hello, first of all, thanks for sharing this library! I know this is an old report, but as I see It is still open and I am experiencing the same issue I will post here.

I am experiencing exactly the same problem as shashi-kiwi. Having my app open in foreground when I call the function:

`
    let simulation;

    const emulateTag = async () => {
        const tag = new NFCTagType4(NFCContentType.Text, "TEST");
        simulation = await (new HCESession(tag)).start();
    }
`

I tap the reader and nothing seems to happen. However if I minimize the app (send it to the background) and tap then, the reader immediately detects the HCE.

Some additional information about hardware and software:

The device used to emit HCE signal:

Android - Samsung Galaxy S9+.
Android version 10

React env:

react: 17.0.2
react-native: 0.67.2
react-native-hce: ^0.1.2

Thanks a lot again, I will update this post If I find a solution on my own!

appidea, 2022-08-21:
Hi guys, recently I had the same problem on one of my devices. After lot of debugging and experimenting with "requireDeviceUnlock" property of aid_list.xml, I finally found the root cause. Possible steps to get rid out of the problem:

- Go to NFC settings in Your phone, more precisely to the list of apps that use HCE. It can be located in different places depending on the Android version/mod, but e.g. in Samsung Galaxy phones with Android 12 You have to go "Settings" -> "Connections" -> "NFC and contactless payments" -> "Contactless payments".
- You will see the list of applications that use HCE. They can be divided into two groups / tabs: "Payment" and "Other".
- Usually You will not be able to deselect any app in "Payment" category, as selection of the payment app works as "radio button" selection style...
- But You should be able to check/uncheck the apps in "Other" menu in "checkbox" manner. Please, uncheck all apps except Yours.
- That's all. Check, if the problem still happens. The application's host apdu service should respond when the NFC component in the mobile device is active (and OFC You previously started NFC tag session in the library). If the problem does not happen now, then apparently one of the installed application conflicts with Your one (in other words, some other app use the same AID as Your app). 

Please, let me know if it helped. If not, probably there is some other / internal limitation that comes from the device/system.

closed 2022-09-06 by appidea

vasilycrespo, 2022-09-14:
Hi! Thank you very much for the support!

I've tested your suggestions, unfortunately, it seems my system was not failing due to this.

However, I am getting closer to identifying the issue. There are two potential reasons.

First of all, I've just installed your example app from:

https://github.com/appidea/react-native-hce/tree/master/example

It seems to work perfectly from there, the reading of the data is consistent, and also it works with the application open / in the foreground.

So I was inspecting any differences with my app, and I noticed that my installed version of react-native-hce was lacking `HCESessionProvider` Not sure if it was a cache issue or something but the yarn/npm version is not bringing the last version from GitHub.

So I installed the library directly from GitHub. And also I change the way of using it to be more like how it is done on the example app.

This gave me more consistent results, at least, 100% of the time reading the card has the correct data.

However, the original issue persists (The hce only works if the app is on background/minimized).

I'm suspecting that the issue might be because my app already had another nfc library used for tag reading, and now the two libs are coexisting on the same app.

https://github.com/revtel/react-native-nfc-manager

I believe that there must be a conflict between the two libraries and probably react-native-nfc-manager is overlapping react-native-hce feature while the app is open.

My next step will be to install a fresh app with the same configuration (react/native version etc) excluding react-native-nfc-manager to see if it works, this might lead to a starting point to identify the exact issue.

vasilycrespo, 2022-09-20:
Hi, just wanted to share that I already solved this, just in case it might be useful for someone else.

 The problem was, that we already had a library installed that uses NFC, the library is from a private repository from a customer, however, the issue was that the library was creating a listener at some point that was using the android NFC service all the time, and with this, it was preventing react-native-hce from working properly (That's why it worked when the app was in the background because the private NFC library had a listener to turn off the service when the app goes to background).

They made some changes to the library exposing a method that allows to stop the listener and now it works perfectly!