appidea/react-native-hce #3: trouble setting up an AID

danielzzz, 2021-01-02:
hi, 

I am having some troubles setting the aid with this library.
If I use any other aid than the one in the example D2760000850101 the app is never selected
this is the debug output for D2760000850102 for example

```
card inserted <Buffer 3b 80 80 01 01>
trying to connect CONNECT_MODE_CARD 2
connected { type: 2, protocol: 2 }
handling tag {
  atr: <Buffer 3b 80 80 01 01>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4'
}
processing ISO 14443-4 tag {
  atr: <Buffer 3b 80 80 01 01>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4'
}
transmitting <Buffer 00 a4 04 00 07 d2 76 00 00 85 01 02 00> 40
transmit response received <Buffer 6a 82> 2

```

any idea why it might happening?

I've trying clearing the cache and reinstalling the app..

thanks cheers
dan

danielzzz, 2021-01-02:
I've seen that there are list of defined aid's as seen here:
https://www.eftlab.com/knowledge-base/211-emv-aid-rid-pix/

...AND also in the source code this lib is activated only on D2760000850101 independently of what you have set in aid_list.xml

cheers, dan

appidea, 2021-01-08:
Hey,

Currently, the library works only with _NFC Type 4 Tag_ - so it's using the AID assigned to this application.
If You would like to support other application type in the library - adding new AID to the xml file is not enough; it's very likely that You will also need to add the support for the applications' APDU commands in `CardService.java`.

Of course, if You would like to extend the library with new functionalities - You are welcome to create the pull request.

closed 2021-01-08 by appidea