appidea/react-native-hce #8: Record Doesn't Send From Samsung Galaxy Note 20

FahdAlShalhoub, 2021-10-06:
Greetings

When activating hce with text content type on a Samsung Galaxy Note 20, the record doesn't transfer when read.

Thank you for all the amazing work, I heavily appreciate all the efforts put in the library, It's been a god-send for me so far.

appidea, 2021-10-18:
Please, ensure the reader supports ISO 7816 Tag mode (NFC Type4 tag). Currently, only this type of NFC Tag is supported, due to the HCE mode limitations on Android.

I have tested this library at least with:

Host:
- Samsung Galaxy S10 

Readers:
- iPhone XR (using NFC Tools application)
- PC (using ACS reader as a hardware via PC/SC protocol).

*Note:* Currently I am working on next major release of this library. It will contain i.e. refactored logic of APDU exchange.

freeridre, 2021-12-05:
Hi @appidea,
I'm struggling with this almost a month, and I could not get it to work. 
So I explain the problem at expo forum, could you help me? [expo forum](https://forums.expo.dev/t/nfc-host-based-card-emulation-hce/57974?u=freeridre)
I noticed that I get the 90 00 response, but my custom message "Hello world" never arrive. I can read the "Hello world" content with my iPhone with NXP tag info, but I do not relly know what's the problem.

closed 2022-09-22 by appidea