appidea/react-native-hce #11: Event Listener To Know if the card has been read/ Can the card be written to

EziamakaNV, 2021-11-11:
Hello,

I wanted to ask if there's an event listener or a way to know if the hce emulator has been read?

And also is it possible for data to be written to the hce?

Hope to hear from you soon.

Thank you

appidea, 2021-11-11:
Hi!
Thank You for pointing out the really good feature which is lacking from current version of `react-native-hce`.

This functionality is to be delivered in the next version. I am working on that.
Follow the repo to get the updates :)

EziamakaNV, 2021-11-11:
Many thanks! I'm trying to implement a p2p app via NFC and I'm really looking forward to this package being able to receive data written to the hce :)

appidea, 2022-09-06:
The functionality have been implemented in master branch, and will go to the next release.

closed 2022-09-06 by appidea