appidea/react-native-hce #12: NFC Link Lost

RebeccaGibbon, 2022-02-14:
Hi!
I included this library in an app I'm using to send data to a PN532.
I am able to detect the emulated tag, however the NFC link is always lost right after I read the nlen value (Finishing service: 0).
I used NDEF Tag Emulator to test the PN532 code and it was able to successfully read the data from the emulated card.
I am using a Samsung Galaxy S10e to test my app.
Do you have any idea why this may be happening?

 Thanks!

appidea, 2022-08-21:
@RebeccaGibbon Thank You for reporting the problem! Sorry for the late response.

Problems like that are always tricky to solve, as it's hard to say where it's the root cause. On the emulator device we just got the information that the link is lost, in this situation I would rather debug the reader device, what kind of error is reported and why the device decided to interrupt the communication (if the interruption does not come from the physical reasons).

Could You provide more information about the problem? 

The library reacts for "READ BINARY" (marked as NLEN in some places) R-APDUs in two cases: when reader wants to read the CC file and when reader wants to read the NDEF message file. Could You please elaborate, on which "READ BINARY" the communication is being interrupted?
Also helpful would be to provide the data which You provide to the emulation (it does not have to be the exactly same string, but helpful would be to know at least the length of it).

appidea, 2022-09-01:
I decided to close the issue due to inactivity. Please, open a new one if needed.

closed 2022-09-01 by appidea