appidea/react-native-hce #14: Issues with 2 installed HCE Apps

SurpriseMF3000, 2022-08-25:
Hello,
im having issues with two installed HCE Apps on one Android Device.
If im calling the HCE function and enter with another phone the NFC field, the HCE phone asks which HCE protocol I want to use.

Example:
Phone 1: Has two Apps with Android HCE installed
Phone 2: should receive the emulated Data from Phone 1
If im calling the HCE function on Phone 1 and enter with Phone 2 the NFC field, Phone 1 asks which HCE protocol I want to use.

How can I declare in my HCE App, that I want to use this one?

Thanks for Help;)

appidea, 2022-08-26:
Hello,

Probably the problem is in two different HCE-enabled applications that are using the same NFC Type4 tag AID (it is, there are more than one application that have the NFC Type4 tag emulation capabilities). In this case, when the device is tapped to the card reader AND the reader is sending the "SELECT APPLICATION" APDU to the card with the given AID; then the phone should ask the user, which of the service to use for handling this specific AID. Unfortunately we cannot change the AID if we want to just emulate the NFC tag, because basically all the (NFC Type4) tags are using one, same AID which is assigned to this application.

Unfortunately there is no possibility to set the importance of one app over another in handling the same-AID operations, but the user, who use the application on the Android phone should be able to tick the "Always use this app to handle this operation" checkbox on the OS-level, when prompted.

Please, let me know if this response answers the question.

SurpriseMF3000, 2022-08-30:
Thanks for the answer;)

closed 2022-08-30 by appidea