appidea/react-native-hce #17: Issues with the new Update

SurpriseMF3000, 2022-09-06:
Dear Appidea,
im having issues with your latest `react-native-hce` Commit.

Error 1:

Error: `TypeError: undefined is not an object (evaluating '_reactNativeHce.NFCTagType4NDEFContentType.Text')`

Code:  
```
const tag = new NFCTagType4({
    type: NFCTagType4NDEFContentType.Text,
    content: "Hello world",
    writable: false
});
```

This one works: `const tag = new NFCTagType4(NFCContentType.Text, "Its working LOLLL Motherfuckers!!!");`

Error 2:

If I´m using the working code, im getting the second error:

`TypeError: undefined is not an object (evaluating '_reactNativeHce.HCESession.getInstance')`

Thanks for Help and this great Repository;)

appidea, 2022-09-22:
Hi, Despite of the fact that new version arrived to master branch, the new version tag has not been created until today. Now the 0.2.0 version is published to npm. Please, try again with installing the most recent version (mind the breaking changes in api) and let me know if it solved Your problem.

SurpriseMF3000, 2022-09-22:
The new version works GREAT;)

I have just one question about the listen function.
```
const listen = async () => {
  const removeListener = session.on(HCESession.Events.HCE_STATE_READ, () => {
    alert("The tag has been read! Thank You.");
  });

  // to remove the listener:
  removeListener();
}
```

If Im reading the emulated HCE message on a different phone via a NFC Reader, Im getting the function triggered on the HCE phone multiple times that the Tag has been read.
This listen function detects when im reading the function but doesn't knows when everything is shared between those two phones.
For example:
This function returns multiple times `alert("The tag has been read! Thank You.");` and if im implementing the `stopSession` function the Card Emulation doesn't works because it terminates too early before everything is shared.

Does `HCESession.Events.HCE_STATE_READ` contains multiple times `HCE_STATE_READ`?

Thanks for an answer and GREAT GREAT GREAT COOL WORK!
Really appreciate it;)

closed 2022-09-30 by SurpriseMF3000