appidea/react-native-hce #19: Express Mode

cfontana0, 2022-09-21:
Hello!
I'm using this library on one of my projects and it works great! thanks for sharing it!
I'd like to implement the HCE feature on iOS, but I understand that it's not possible, right?

Also, I noticed that Apple has an express mode: With Express Mode, you can use some of your cards, keys, and passes in Apple Wallet without waking or unlocking your device, or authenticating with Face ID, Touch ID, or your passcode. You might even be able to use your card, pass, or key when your device needs to be charged. https://support.apple.com/en-us/HT212171

Do you know if that is possible to implement using this library on Android? maybe it's a feature that is not development, but I'm happy to contribute if you see it possible to implement it.

Thanks!
Charlie Fontana

appidea, 2022-09-22:
Hello!
Thanks for Your appreciation. I am happy that You find this work meaningful :)

Regarding that Apple™ Wallet features, I am familiar with the capabilities. Thanks for a good point! These functionalities are in fact close to the HCE services, but I think not quite fit to this specific project. 
The general goal of this library is, apart from simulating the NFC Tag (that would in the future become just `one of the many` functions), to create a RN bridge to connect with NFC chip in a "low-level manner". But this is generally a good idea for a separate library, that would be definitely worth to see in the community.

Back to the HCE situation in iOS, the low-level HCE API (as on Android) is not available at the time, thus there is no plan for development of any iOS functionality here. All that we can do is... just to hope that Apple will decide to open HCE api in some future. If it would happen, I will surely add the iOS support to the roadmap.

closed 2022-09-22 by appidea

cfontana0, 2022-09-22:
Hey! ok! Thanks for sharing your thoughts