appidea/react-native-hce #22: Unable to Send NDEF message from android 10 to android 12

vasilycrespo, 2022-10-18:
Hello, first of all, thank you very much for a great job with this react native library.

I am experiencing the following issue:

I am able to use the HCE library on both android 10 and 12, which are the devices I am testing with.

Both are able to send the HCE message, however android 12 is unable to read the message from android 10.

For example, I can send an NDEF message from android 10 to another android 10 device as a reader.

However, if a try to send the NDEF message from android 10 to a Samsung S22+ with android 12, the reader is unable to receive the NDEF messages, there are no errors on any of the two sides. There is just a small bubble on the HCE side (android 10) which states that there are no apps available to receive the tag.

As a side note I am using this library as a reader:

https://github.com/revtel/react-native-nfc-manager

Also, I tested sending the HCE NDEF message with another google play store library and the message is properly sent from android 10 to 12 with this one:

https://play.google.com/store/apps/details?id=com.maxsoft.ndeftagemulator&hl=en&gl=US

Just wanted to know I anyone has experienced something similar or has any clue of what could be happening. Thanks in advance!

appidea, 2023-04-04:
@Zardozz Many thanks! It could be the most wanted answer for this question of random behaviour... I will definitely play around with making the CC less and if it will help with compatibility, I will apply the appropriate changes to the next release.

Zardozz, 2023-04-05:
As per the stackoverflow question, it seems some success on reducing the CC size to `7FFF` has some success as this is a more logical max size of 32K to match physical Tags.

PraneethGunas, 2023-08-06:
Any update on this?

IZZAGOLD, 2023-10-03:
Any update on this?

yasiragi, 2023-10-05:
I'm also looking an update on this, since I'm able to emulate but not able to read the content on another android device scanning using react-native-nfc-manager.

jwh-hutchison, 2025-03-03:
any update on this?

papidb, 2025-04-11:
Any update on this please?

appidea, 2025-07-30:
#40 should fix this. New version including this fix is now out.

closed 2025-07-30 by appidea