appidea/react-native-hce #24: Cannot read property 'on' of undefined

0xcD3v, 2023-03-29:
I tried to run this code, and i got cannot read property 'on' of undefined could you help me, am i doing something wrong? 

`````````
import {
  HCESession,
  NFCTagType4NDEFContentType,
  NFCTagType4,
  HCESessionProvider,
} from 'react-native-hce';

function App() {
  let session;

  const startSession = async () => {
    const tag = new NFCTagType4({
      type: NFCTagType4NDEFContentType.Text,
      content: 'Hello world',
      writable: false,
    });

    session = await HCESession.getInstance();

    console.log('SESSION' + JSON.stringify(session));
    session.setApplication(tag);
    await session.setEnabled(true);
  };

  const stopSession = async () => {
    if (session) {
      await session.setEnabled(false);
    }
  };

  useEffect(() => {
    startSession();
    const removeListener = session.on(HCESession.Events.HCE_STATE_READ, () => {
      ToastAndroid.show('The tag has been read! Thank You.', ToastAndroid.LONG);
    });

    return () => {
      stopSession();
      removeListener();
    };
  }, []);

  return (
    <HCESessionProvider>
      <TouchableOpacity onPress={startSession}>
        <Text>EMULATE TAG CARD </Text>
      </TouchableOpacity>
    </HCESessionProvider>
  );
}

appidea, 2023-04-04:
Looks like You call ``on`` to early. In Your current implementation You should put the entire logic from useEffect to some async function and put await keyword before startSession call: ``await startSession();``. 

Anyways, looks like a problem with implementation, not with the library itself.

scyther, 2023-04-05:
I am using the useDataLayer Hook of example code but still eventData is always undefined

0xcD3v, 2023-04-13:
you right  i wrote this  new code and it works 👍 
``````````````````````````
import React, {useState, useEffect} from 'react';
import {View, Text, Pressable, ToastAndroid} from 'react-native';
import {
  HCESession,
  NFCTagType4NDEFContentType,
  NFCTagType4,
} from 'react-native-hce';

const App = () => {
  let session;

  const startSession = async () => {
    alert('emulating');
    const tag = new NFCTagType4({
      type: NFCTagType4NDEFContentType.Text,
      content: 'Hello world',
      writable: false,
    });

    session = await HCESession.getInstance();
    session.setApplication(tag);
    await session.setEnabled(true);
    listen();
  };

  const stopSession = async () => {
    await session.setEnabled(false);
    alert('bye');
  };

  const listen = async () => {
    const removeListener = session.on(HCESession.Events.HCE_STATE_READ, () => {
      ToastAndroid.show('The tag has been read! Thank You.', ToastAndroid.LONG);
    });

    // to remove the listener:
    removeListener();
  };

  return (
    <View style={{flex: 1, justifyContent: 'center', alignItems: 'center'}}>
      <Pressable style={{padding: 20}} onPress={startSession}>
        <Text>Start</Text>
      </Pressable>
      <Pressable onPress={stopSession}>
        <Text>Stop</Text>
      </Pressable>
    </View>
  );
};

export default App;
``````````````````````````

closed 2023-04-13 by 0xcD3v