appidea/react-native-hce #26: HCE_STATE_READ event is fired multiple times on tag reading

buenaondalab, 2023-05-17:
I'm trying to emulate a NFCtagType4 with an Android device.
I'm able to read the tag but HCE_STATE_READ event is fired 3 times.
Is that the desired behaviour?
I fixed it locally in respondRead method inside NFCTagType4 class:

``` java
if (realLength == slicedResponse.length) {
      this.hceModel.getLastState()
                   .setValue(HceViewModel.HCE_STATE_READ);
    }
```
Could it be a valid solution to avoid multiple firings?

neilanthonyte, 2024-11-02:
Same here, it keeps bumping up and the only way to fix it is to close the app to reset the session . . even if the component is unmount and re-mount it triggers a lot

neilanthonyte, 2024-11-02:
any solution on how to fix this?

```
let removeListener = session.on(HCESession.Events.HCE_STATE_READ, async () => {
               /** LISTENER KEEPS TRIGGERING MULTIPLE TIMES, LIBRARY ISSUE
                *  CREATE A FLAG TO EXECUTE THE ALERT ONCE
                *  LISTENER GETS CLEARED ONLY WHEN THE APP IS CLOSED
                **/                
                if(execFlag === false) {
                    execFlag = true;

                    Alert.alert(
                        'Success',
                        `NFC connection successfully completed.`,
                        [{text: 'Ok', onPress: async () => {
                            await session.setEnabled(false);
                            setTimeout(() => {
                                dispatch(showTagging({show: false, payload: null}));
                            }, 300)
                        }}]
                    );
                }
            });

            removeListener();
```
            
            listener is not removed even if the component is unmounted . .