appidea/react-native-hce #29: Build fails on the latest version of React Native v0.76.7

robertclarkson, 2023-08-03:
Fresh project installed from React Native v0.76.7

```
npx react-native run-android
```
```
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':react-native-hce'.
> Failed to notify project evaluation listener.
   > Could not create task ':react-native-hce:compileDebugAndroidTestKotlin'.
      > Cannot use @TaskAction annotation on method AbstractKotlinCompile.execute() because interface org.gradle.api.tasks.incremental.IncrementalTaskInputs is not a valid parameter to an action method.
   > KotlinJvmAndroidCompilation with name 'debugAndroidTest' not found.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
error Failed to install the app. Command failed with exit code 1: ./gradlew tasks FAILURE: Build failed with an exception. * What went wrong:
A problem occurred configuring project ':react-native-hce'.
> Failed to notify project evaluation listener. > Could not create task ':react-native-hce:compileDebugAndroidTestKotlin'. > Cannot use @TaskAction annotation on method AbstractKotlinCompile.execute() because interface org.gradle.api.tasks.incremental.IncrementalTaskInputs is not a valid parameter to an action method. > KotlinJvmAndroidCompilation with name 'debugAndroidTest' not found. * Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights. * Get more help at https://help.gradle.org BUILD FAILED in 2s
> Task :gradle-plugin:compileKotlin UP-TO-DATE
> Task :gradle-plugin:compileJava NO-SOURCE
> Task :gradle-plugin:pluginDescriptors UP-TO-DATE
> Task :gradle-plugin:processResources UP-TO-DATE
> Task :gradle-plugin:classes UP-TO-DATE
> Task :gradle-plugin:jar UP-TO-DATE
> Task :gradle-plugin:inspectClassesForKotlinIC UP-TO-DATE Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0. You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins. See https://docs.gradle.org/8.0.1/userguide/command_line_interface.html#sec:command_line_warnings
5 actionable tasks: 5 up-to-date.
```

roeicohen, 2023-08-05:
This happens to me as well.

appidea, 2023-08-22:
The problem is that in the freshly-generated project does not define kotlin version. Add this to ``android/build.gradle`` - take care to define at least v1.6. I will be investigating further on that matter.

```
buildscript {
    ext {
        ...
        kotlinVersion = "1.6.20"
        ...
    }
}
```

closed 2025-07-20 by appidea