appidea/react-native-hce #34: Can I open a specific application when the Emulating device is read.

aasharck, 2023-10-25:
First of all, thanks a lot for this package! Really appreciate it.

I am a noob to Mobile app development, so please forgive me if I am wrong. So, when I emulate a phone using this package, I want the reading device/phone to open a specific app. Can we do this using this package? Is this handled here or is it handled by the reader? Because I saw this article from Android official documentation - https://developer.android.com/guide/topics/connectivity/nfc/nfc?authuser=1#tag-dispatch

If it's possible please let me know! Thanks

appidea, 2023-10-31:
Opening of the reader app is handled on the reading device. Android picks the proper app by AID. So if You want to be sure that the tag will be opened with the proper reader app on the target device:

- Your reader app should bind to the specific AID
- Your sender (HCE) app should send the same AID as reader app expects

Currently the react-native-hce handles only one AID that is assigned to "NFC Tag Type 4".

**Be careful to not pick the reserved AID!**

AID is specified in the manifest file.

aasharck, 2023-10-31:
Makes sense. Thank you so much!

closed 2023-10-31 by aasharck