appidea/react-native-hce #35: Fixed: Unable to send NDEF message

IZZAGOLD, 2023-11-03:
Hello, I am the owner of Pixel 4a and 7a (android 13 and 14). I couldn't emulate the tag.
I found the solution in the following: in the NFCTagType4.java file
change
public final byte[] ndefDataBuffer = new byte[0xFFFE];
on
public final byte[] ndefDataBuffer = new byte[0x7FFF];

change
public final byte[] ccDataBuffer = new byte[15];
on
public final byte[] ccDataBuffer = new byte[] {
       0x00, 0x0f, // CCLEN
       0x20, // Mapping Version
       0x00, 0x3b, // Maximum R-APDU data size
       0x00, 0x34, // Maximum C-APDU data size
       0x04, 0x06, // Tag & Length
       (byte)0xe1, 0x04, // NDEF File Identifier
       (byte)0x7F, (byte)0xFF, // Maximum NDEF size
       0x00, // NDEF file read access granted
       (byte)0xff, // NDEF File write access denied
   };

delete
setUpCapabilityContainerContent and its call
I don't use tag rewriting and I don't care about the writable property.
Please improve these points. Thank you

dmolinamesa01, 2024-06-14:
Hi, I'm triying to fix this problem on my react native app doing your fix. However I can't find this file in my project. How can I fix it then?

dmolinamesa01, 2024-06-14:
Hi, I'm facing this problem. When I scan the tag with NFC tools always says that has not readonly able. I have also tried to get the content message working with react-native-nfc-manager but its impossible to get the content message. I couldn't solve it running your fix. Where did you get this info to set that on this file? Thanks

papidb, 2025-04-11:
Re ask

PR #40 (2025-07-20): Fix emulated Capability Container to achieve Android compatibility
Dzień dobry!

It seems like recent Android phone versions don't like the Capability Container values as they are currently emulated within this project.

This PR:

1. Adjusts the Capability Container to the correct size of 0x0F (resolve #37).
2. Lowers the values indicated by fields mentioned below in order to achieve greatest possible interoperability (resolve #38 resolve #35 resolve #31).
   * MLe (bytes); Maximum RAPDU data size
   * MLc (bytes); Maximum CAPDU data size
   * NDEF File Control TLV - Maximum NDEF file size

Th [... cut]
closed 2025-07-20 by appidea