appidea/react-native-hce #36: Add suport for IOS

29Hido, 2024-03-22:
Hey

Since IOS 17.4, Apple added an [official support for HCE](https://developer.apple.com/fr/support/hce-payment-transactions-in-payment-apps/), is it planned to add an interface for it in react-native-hce ?

Thank you.

ailtonaires, 2024-05-24:
I'm waiting for it

NorbertBodziony, 2024-09-02:
would love to see it !

neilanthonyte, 2024-11-04:
looking forward to use it

QcFe, 2025-07-22:
any update on this? 😬

icedevml, 2025-09-13:
Apart from that library, I've just released a new project which enables to perform low level HCE on both Android and iOS.
https://github.com/icedevml/react-native-host-card-emulation

icedevml, 2025-09-14:
@appidea Honestly I was really considering to add the iOS piece to your project as a PR but there were a few circumstances that stopped me on this.

I was in need to have it super low level so to be able to implement stuff like a custom door lock protocol for a customer. I also wanted to directly receive C-APDUs and R-APDUs on the JavaScript side, which is a super huge refactor already. Also, my PoC module was in the React Native "New Architecture", so the codebase is not fully compatible. I'm not an expert on React Native, just trying to do my job on a single, very specific project, so honestly the module development was quite of a struggle for me :P

I think that your original project is still very worthful if somebody wants to use Android to emulate NDEF or just quickly implement some other high-level protocol within. On the flip side, my library should allow to implement almost anything, but is not straightforward to use without some NFC/ISO 14443/ISO 7816 expertise. So just we just have two separate use cases here.