appidea/react-native-hce #38: Can't read an NFC tag

Henry-E, 2024-07-02:
Hi,

I'm testing out this library using expo. I've added all the necessary lines to the manifest and the aid_list.xml file.

But I'm still having a lot trouble getting the example app working. Clearly something is working, the app doesn't crash when it attempts to turn on HCE and when I have two different HCE apps running, the android UI pops up and asks which app should be responsible for generating the tag. The only issue is that the other app actually generates a tag that can be read and the example doesn't. 

The most frustrating part is the difficulty in debugging or trying to narrow down what's going wrong. It's using all the same example code and the manifest + aid_list have been added / updated. The HCE logger detects when the HCE is turned on and off, it detects when another tag is read. But it never actually generates the tag or the data from it. 

In the event when there are two apps with HCE on the phone both active, the android UI asks to choice between them. When I choose the other app the nfc reader on the other phone detects the tag. When I choose the react-native-hce example app the UI for tapping the phone against a card reader shows up but the tag reader can never detect the generated tag. In the event that only the react-native-hce app is active, then no android options show up and the tag is still never read.

For reference here's the working app, which at least proves that my devices can do HCE with other devices correctly. <https://github.com/MichaelsPlayground/NfcHceNdefEmulator/tree/master>.

But I have no idea
- how to debug what's going wrong from here
- how to fix the issue

Appreciate any help possible

Henry-E, 2024-07-03:
My best guess is that it's linked to the array size issue mentioned on SE and another issue. https://stackoverflow.com/questions/75932985/why-does-data-transfer-over-nfc-with-hce-only-work-with-android-lower-12

Is it at all possible to merge or test this change to reduce the size of the max data?

I've been trying to run the code from this repo to test changes locally but keep running into errors and stuff that's out of date.

appidea, 2024-07-03:
Hi Henry,

I am going to publish the new version of this library soon, taking into account part of the issues that are on the board. The issue You've mentioned is one of these. There is generally a lot of legacy stuff to be changed or removed.

Please follow the repo to be notified about the update.

Br,
mat

Henry-E, 2024-07-04:
Awesome, thanks for the update mat.

I can possibly help to test with integrating the expo plugin file that wodin wrote once the tests work again. I'm still a bit of a noob at this react native and gradle stuff.

PR #40 (2025-07-20): Fix emulated Capability Container to achieve Android compatibility
Dzień dobry!

It seems like recent Android phone versions don't like the Capability Container values as they are currently emulated within this project.

This PR:

1. Adjusts the Capability Container to the correct size of 0x0F (resolve #37).
2. Lowers the values indicated by fields mentioned below in order to achieve greatest possible interoperability (resolve #38 resolve #35 resolve #31).
   * MLe (bytes); Maximum RAPDU data size
   * MLc (bytes); Maximum CAPDU data size
   * NDEF File Control TLV - Maximum NDEF file size

Th [... cut]
closed 2025-07-20 by appidea